tue-robotics/tue_robocup #897: PR robot_smach_states

PetervDooren, 2019-09-20:
- [ ] CMakelists.txt
- [ ] Makefile
- [ ] navigate_to_place_expl.py
- [ ] console.py
- [ ] designator_iterator.py
- [ ] .detect_face.py.swp
- [ ] human_interaction/__init__.py
- [ ] find_person_in_room.py
- [ ] give_directions.py
- [ ] human_interaction.py
- [ ] show_image.py
- [ ] manipulation.py
- [ ] control_to_pose.py
- [ ] find.py
- [ ] guidance.py
- [ ] navigate_to_observe.py
- [ ] navigation.py
- [ ] rise.py
- [ ] startup.py
- [ ] designators/__init__.py
- [ ] designators/checks.py
- [ ] designators/core.py
- [ ] designators/ed_designators.py
- [ ] designators/string_manipulation.py
- [ ] designators/utility.py
- [ ] util/dummy_state.py
- [ ] util/startup.py
- [ ] world_model.py
- [ ] test_give_directions.py
- [ ] test_grasping.py
- [ ] test_handover_to_human.py
- [ ] test_navigate_to_observe.py
- [ ] test_perception.py

PetervDooren, 2019-09-20:
@LarsJanssenTUe shall I move find_people_in_room.py and find_person_in_room.py to a different PR? Than I can strike them off the list

LarsJanssenTUe, 2019-09-21:
@PetervDooren, yes I think that that would be a good idea

closed 2019-10-09 by PetervDooren