tue-robotics/tue_robocup #898: Standardize python docstrings

MatthijsBurgh, 2019-09-21:
Many style of docstring are used. This should be consistent. As automatically generated documentation is also an issue. The docstring should be compatibel with the documentation generator we will be using. Preferably one which has automatic linking

Rayman, 2019-09-21:
This package seems really nice: https://github.com/mikepurvis/catkin_tools_document. Builds documentation of the whole workspace with rosdoc_lite.

LoyVanBeek, 2019-09-22:
Another very useful thing: add type hints! If you use PyCharm, that'll also help with completion

LarsJanssenTUe, 2020-03-31:
@MatthijsBurgh I think this issue can now be closed? Although a new one might be opened were we keep tabs on which repo's still need documentation?

MatthijsBurgh, 2020-03-31:
It might be good to have some examples, so people will use the good styling.

closed 2021-01-27 by PetervDooren