tue-robotics/tue_robocup #939: Find people finds people outside the assigned area

LarsJanssenTUe, 2019-12-03:


PetervDooren, 2020-06-21:
Is there anymore info on this issue. How do we reproduce this issue

MatthijsBurgh, 2020-09-25:
https://github.com/tue-robotics/tue_robocup/blob/22e6874f403c21814005ce4808ed722755d8edc9/robot_smach_states/src/robot_smach_states/human_interaction/find_people_in_room.py#L187-L198

Is this the issue?

LarsJanssenTUe, 2020-10-06:
> Is there anymore info on this issue. How do we reproduce this issue

See the video of the finals in Sydney. No people outside of the arena should be detected (I know that there was even some kind of buffer zone inside the arena in which no people should be detected), but they were.

MatthijsBurgh, 2020-10-06:
Yes I know, but it the suggested code the origin of the problem.

closed 2021-01-27 by PetervDooren