tue-robotics/tue_robocup #944: [Story] Demoability

LarsJanssenTUe, 2019-12-03:
The robot should be workable for the demo team. What is needed for this?

**Changes that need to be made to the robot**
- [ ] The start-up procedure should be as robust as possible (i.e. get rid of virtual box!)
- [ ] Some how the robot should always be in a demoable state (make a separate user?)

- [ ] (Un)pack manual (external demo)
- [ ] Hero manual 
  - [ ] How to start the robot
  - [ ] How to start the demo
    - [ ] The difference in procedure between demo intern and demo extern
  - [ ] What commands can be given to the robot (both through speech and through telegram)
- [ ] Presentation 
  - [ ] Written tekst
  - [ ] Presentation

LarsJanssenTUe, 2020-03-31:
@MatthijsBurgh progress report?

MatthijsBurgh, 2020-03-31:
I think the checked boxes show the status. The laptop is readyish. I am not sure anymore.

MatthijsBurgh, 2020-08-18:
Demo laptop is at 18.04

PetervDooren, 2021-01-27:
todo: change to a discussion when this is possible again

PetervDooren, 2022-05-16:
Some notes on the demo software as of 16-05-2022,

- [ ] Make sure we can press and release the emergency button without having to restart everything.
  - [ ] Demo intern
  - [ ] Demo extern (gmapping will probably pose a problem.)
- [ ] Add desktop shortcuts for Rviz and the gui server.
- [ ] Place action is currently broken
- [ ] Fix arm control in gui. Make it easier to reset the robots pose. 
- [ ] The robot avoids its own arm while navigating.

MatthijsBurgh, 2022-05-16:
Gmapping will remain an issue, because the pose will be reset.
Also AMCl, normal localization, will require a new initial pose. As again the pose/odom is reset.
Robot avoiding its own arm could be an issue caused by the navigation cloud. We do not do self filtering there at the moment. This also caused issues in Sydney, while carrying trashbag or similar. Though it should filter in the costmap for everything in the footprint.
Rviz is routed to the global planner. The replanning is implemented in the robot_smach_states, so that is not a ROBOT interface.