tue-robotics/tue_robocup #949: arm.send_joint_goal no longer works

PetervDooren, 2019-12-09:
While testing arm.send_joint_goal() no longer seems to work in the hero-console and possible in challenges as well. The error message is:

Dropping all 1 trajectory point(s), as they occur before the current time.
Last point is 0s in the past.

PetervDooren, 2019-12-14:
Issue does not seem to be reproducable. robot seems to work fine again.
For now we close the issue. If this happens again in the future we should look into this

closed 2019-12-14 by PetervDooren

MatthijsBurgh, 2019-12-14:
@PetervDooren did you try publishing the message manually like we did last week?

PetervDooren, 2019-12-15:
I tried using the hero-console but the robot responded as normal. It could be that last week was just a freak occurance