tue-robotics/tue_robocup #973: Move Hero with controller

LarsJanssenTUe, 2019-12-17:


JosjaG, 2020-03-07:
Two nodes are required to achieve the setup described in the tmc manual: the `interactive_teleop_joy` node of tmc and the standard `joy_node` of the joy package:
``` 
roslaunch hsrb_interactive_teleop_config hsrb_interactive_teleop_joy.launch
export ROS_NAMESPACE=/hsrb
rosrun joy joy_node 
```

Then the problem arises that the interactive teleop node publishes to standard Toyota topics, and not out own (so hsrb stuff instead of hero), for example: it publishes to `/hsrb/command_velocity` whilst in our case, we want that to be `/hero/base/references`.

JosjaG, 2020-03-07:
We could develop our own node that subscribes to the joy topic, then we can choose ourselves what each button does. This might take some more time but it does not require that we remap all the standard Toyota topics (using the interactive_teleop node does require this).