tue-robotics/tue_robocup #988: Fix docstrings syntax for documentation

MatthijsBurgh, 2020-01-19:
### cross-references
https://stackoverflow.com/questions/7753805/sphinx-automodule-how-to-reference-classes-in-same-module and https://www.sphinx-doc.org/en/master/usage/restructuredtext/domains.html#cross-referencing-syntax

Use 
```
:class:'.A'
``` 

but with backticks instead of quotes

_Originally posted by @LoyVanBeek in https://github.com/tue-robotics/tue_robocup/pull/978_

LoyVanBeek, 2020-01-20:
For the documantation generator, it's also important to have a newline between the docstring summary and the first `:param` stuff. So 

```
This method does methody-things

:param thing: A thing to method on`
``` is OK

but 

```
This method does methody-things
:param thing: A thing to method on`
``` is not ok.

LarsJanssenTUe, 2020-03-31:
@MatthijsBurgh I think this can be closed?

MatthijsBurgh, 2020-03-31:
Same here, we should come up with an example for correct docstring styling.

closed 2021-01-27 by PetervDooren