tue-robotics/tue_robocup #999: Storing wanted arm properties

LarsJanssenTUe, 2020-02-02:
Currently there is no way to document what you can put into the structure and what values mean, no way to do any form of checking basic sanity until you try to construct an arm deep down in the robot rather than while you construct a properties object, no way to find all uses if you ever need to change the interface, no way to attach a method to it (and likely the list is not complete).

LoyVanBeek, 2020-02-02:
Maybe we can define some `ArmRequirements` type that each (manipulation) state can set for itself. In the `__init__` of that state, those are gathered piece-by-piece. 

Another option might be to go through the state machine and check for arm requirements and gather them tat way. 
This is getting out of hand though for the 1-armed robot we currently mostly use, but is indeed quite generic.

LoyVanBeek, 2020-02-03:
Another option might by to extend our `ArmDesignator` base-class with ways to extend the requirements to an arm, so that when the time to resolve the designator comes, all requirements have been gathered. 
E.g.:

```python
arm_des = UnoccupiedArmDesignator(...)

class BlaBlaState(...):
    def __init__(robot, arm_designator):
        arm_designator.require(has_pose='pointing')

    def execute(...):
        arm = arm_designator.resolve()

class FooState(...):
    def __init__(robot, arm_designator):
        arm_designator.require(has_pose='foo_pose')

blabla = BlaBlaState(arm_des)
foo = FooState(arm_des)
```

For different designators, we have different states it's passed through so different requirements

alberth, 2020-02-04:
The comment by @LarsJanssenTUe is pulled out of context, my argument in https://github.com/tue-robotics/tue_robocup/pull/980#issuecomment-576255355 was that code like
```
      arm_properties={"required_trajectories": ["prepare_grasp"],
                      "required_goals": ["carrying_pose"],
                      "reequired_gripper_types": [arms.GripperTypes.GRASPING]},
```
is complete magic, I suggested to have a class with the same fields, so you have a place for documenting, and room for any processing on the supplied arguments, for example typo checking ;p

alberth, 2020-02-04:
Eg you could create an instance, and visit all used smach state machines with it to populate it with requirements of the state machines.

LarsJanssenTUe, 2021-04-13:
Fixed in  #1016 and #1092

closed 2021-04-13 by LarsJanssenTUe