tue-robotics/tue_robocup #1012: [Story] Speech Recognition

ar13pit, 2020-03-07:
- [ ] Add closed grammar functionality to yapykaldi (Refer to zamia speech JSGF parser and adaptation)
- [ ] Add class based API for kaldi_model_builder->kaldi_decoder->microphone in yapykaldi (This will be wrapped as a HMI client).
- [ ] Decide where parsing of the output of yapykaldi into semantics must be done. In yapykaldi or in the ros wrapper.
- [ ] Create a ros HMI client wrapper for yapykaldi

LoyVanBeek, 2020-03-31:
I'll take up the 
>  Create a ros HMI client wrapper for yapykaldi (yapykaldi_ros)

ar13pit, 2020-03-31:
> I'll take up the
> 
> > Create a ros HMI client wrapper for yapykaldi (yapykaldi_ros)

Great. There is one feature that is needed there though, caching of the grammar string if the same grammar was sent in the previous HMI query as compiling the speech model all the time will be very expensive.

LoyVanBeek, 2020-03-31:
Allright. I've spent most my time setting up the dependencies and I need to be on 18.04 I think though, so I'll continue next week.

LarsJanssenTUe, 2020-03-31:
> Allright. I've spent most my time setting up the dependencies and I need to be on 18.04 I think though, so I'll continue next week.

Needing to be on 18.04 would be undesired if not explicitly necessary. Why is this needed @ar13pit ?

ar13pit, 2020-03-31:
No its not needed. I wrote that in the README as I explicitly tested it on 18.04 but it will work on 16.04 as well as along as CUDA is not being used and gcc7 is used.

LoyVanBeek, 2020-04-01:
Hmm, kaldi complained about needing CMake 3.12 or somethingh while by 16.04 box has 3.5 and could not get a higher version installed yet.

ar13pit, 2020-04-01:
Did you try running the command `tue-get install python-yapykaldi` ? It should install all the dependencies, build kaldi and yapykaldi.

However, if you don't want to do that go with `pip2 install --user cmake`. You won't have to spend time building it from source.

LoyVanBeek, 2020-04-18:
@ar13pit is https://github.com/gooofy/zamia-speech/#model-adaptation what you meant with "Zamia speech JSGF"?

ar13pit, 2020-04-18:
Yup and that section is a huge pile of shit.

LoyVanBeek, 2020-04-19:
@ar13pit 
https://github.com/tue-robotics/yapykaldi_ros/pull/1 and https://github.com/tue-robotics/yapykaldi/pull/1 can be reviewed

LoyVanBeek, 2020-04-21:
>  Decide where parsing of the output of yapykaldi into semantics must be done. In yapykaldi or in the ros wrapper.
I currently have this in the ROS wrapper and I think that makes the most sense too. 

Same for the converting to JSGF format: I'd say that happens in the ROS wrapper and the JSGF grammar is passed to `Asr.recognize` or `Asr.start`.

ar13pit, 2020-04-21:
I actually getting rid of JSGF completely as that is also an intermediate format. Instead I'll keep NLTK grammar object as that is pretty close to our grammar parser. So this object is passed onto `Asr.recognize` and a new ASR model is created.