tue-robotics/tue_robocup #1089: Nodes in scripts or nodes folder

MatthijsBurgh, 2021-02-25:
As @Timple mentioned in https://github.com/tue-robotics/node_alive/commit/6b015b71d4a8e72a68fca4fbbee725294ac7fc65#commitcomment-47542367, nodes should actually be placed in a nodes folder instead of scripts.

Do we want to apply to this convention or keep everything in scripts? In we do want to apply, force a refactor or converge slowly?

ar13pit, 2021-03-16:
Actually neither. In setuptools the convention is to have it generated automatically by referring to a function in the library. So until then I think we can stick on with scripts and should slowly proceed to using setuptools

Timple, 2021-03-22:
> In setuptools the convention is to have it generated automatically

Cool! Do you have a reference for this?

ar13pit, 2021-03-23:
> > In setuptools the convention is to have it generated automatically
> 
> Cool! Do you have a reference for this?

Yes its called entry points, through console_scripts (https://setuptools.readthedocs.io/en/latest/userguide/entry_point.html)

MatthijsBurgh, 2022-05-30:
Conclusion:
>In setuptools the convention is to have it generated automatically by referring to a function in the library. So until then I think we can stick on with scripts and should slowly proceed to using setuptools

closed 2022-05-30 by MatthijsBurgh