tue-robotics/tue_robocup #1109: Left and right arm rostopics on hero

LarsJanssenTUe, 2021-05-01:
```
/hero/left_arm/grasp_precompute/cancel
/hero/left_arm/grasp_precompute/feedback
/hero/left_arm/grasp_precompute/goal
/hero/left_arm/grasp_precompute/result
/hero/left_arm/grasp_precompute/status
/hero/left_arm/gripper/action/cancel
/hero/left_arm/gripper/action/feedback
/hero/left_arm/gripper/action/goal
/hero/left_arm/gripper/action/result
/hero/left_arm/gripper/action/status
/hero/right_arm/grasp_precompute/cancel
/hero/right_arm/grasp_precompute/feedback
/hero/right_arm/grasp_precompute/goal
/hero/right_arm/grasp_precompute/result
/hero/right_arm/grasp_precompute/status
/hero/right_arm/gripper/action/cancel
/hero/right_arm/gripper/action/feedback
/hero/right_arm/gripper/action/goal
/hero/right_arm/gripper/action/result
/hero/right_arm/gripper/action/status
```
These topics probably exist because there are still subscribers for it:

```
(hero1.local-core) lars@athome-lars $ rostopic info /hero/left_arm/grasp_precompute/goal
Type: tue_manipulation_msgs/GraspPrecomputeActionGoal

Publishers: None

Subscribers: 
 * /hero/manipulation_bridge (http://hero1.local:42990/)

(hero1.local-core) lars@athome-lars $ rostopic info /hero/left_arm/gripper/action/cancel
Type: actionlib_msgs/GoalID

Publishers: None

Subscribers: 
 * /hero/gripper_bridge (http://hero1.local:33128/)

```

PetervDooren, 2021-05-03:
Yes this should be a simple matter of removing the lines from hero_bridge

PetervDooren, 2021-05-04:
I think this was with an older version of the software. Current implementation does not have these

closed 2021-05-04 by PetervDooren