tue-robotics/tue_robocup #1159: Depth Image zero when to close

MatthijsBurgh, 2022-01-03:
**For I now I only experienced it in simulation**

Depths that are too close or too far away are zero.

This gives problems in segmentation and object fitting when HERO is too close to the object.

PetervDooren, 2022-01-04:
That seems to me like a problem in the segmentation and object fitting algorithms. These algorithms  do not detect the circumstances in which they perform well, which can lead to errors.

MatthijsBurgh, 2022-01-04:
The cause is just that we don't get the data from the sensor that we would like to have. There is just no depth known for these pixels. So there just isn't any data.

PetervDooren, 2022-01-04:
That is just the way the camera works, isnt it? Not really anything we can do about it. Only thing we can do is use the data we do have

MatthijsBurgh, 2022-01-04:
The question is Does the robot work the same as the simulator. In that case we need to make sure that we are never to close.

PetervDooren, 2022-01-04:
In that case we should be aware that our algorithms dont work when we are too close. Not "never be too close" because I can see that coming back to bite us