tue-robotics/tue_robocup #1175: Robocup 2022: Take out the garbage

PetervDooren, 2022-05-07:
Current state:

Todo:

P-ict0, 2022-05-07:
# Current state: 
Robot correctly drives and positions itself in front of the first trashbin, attempt to grab the trash, lifts it and force drives backwards (hardcoded). Drives to the drop off area and crashes

# Todo: 
- (Josja) Fix the command "hero-challenge-take-out-the-garbage" (has to be rosrun manually when using physical robot).
- (Rodrigo) Try to improve control_to_trash_bin.py
- ...

JosjaG, 2022-05-07:
I just did a complete successful run of the challenge, so the main flow of the challenge is working!! :partying_face:

P-ict0, 2022-05-17:
Just realized the  ```radius```  parameter in  ```control_to_trash_bin.py``` (also called in  ```pick_up.py```) is irrelevant since in state  ```"GO_BIN"```  we tell hero to go in front of the trash can. 

We either have to change  ```"GO_BIN"```  to be able to chose what distance should hero position itself to the trash can, or rewrite ```control_to_trash_bin.py``` with the default distance hero positions itself when ```NavigateToSymbolic(...)``` is called

MatthijsBurgh, 2022-05-24:
force_drive is broken in combination with this challenge. @P-ict0 is taking a look before next dev evening.

PetervDooren, 2022-06-18:
Some shortcomings of the current implementation:
Lots of magic numbers to determine the pose of the robot with respect to the trashbin. I think we can do better than this. It would also lower tuning time when adjusting to a new bin. 
Trashbin size was incorrect in the worldmodel https://github.com/tue-robotics/ed_object_models/pull/100 should fix this
Currently the arm moves down until it feels a force or it reaches a timeout. We can also make it so that we can adjust the distance at which we close the gripper anyway.