tue-robotics/tue_robocup #1255: Generic smach_states

MatthijsBurgh, 2022-11-08:
- [ ] where is this:
  - [ ] smach_state_name
- [ ] hand me that:
- [ ] set the table:
- [ ] clean the table:
- [ ] carry my luggage:
- [ ] find my mates:
- [x] serve breakfast: nothing
- [ ] receptionist:
- [ ] take out the garbage:
- [ ] restaurant:
- [ ] final:

shanki98, 2022-11-08:
carry my luggage seems to be taken from 2019, with generic smach_states already taken.

shanki98, 2022-11-22:
Challenge receptionist has findpeople.py. This script holds the skill to store people information as a struct within a room. I don't know if this has been made as a generic smach state as we have smach states to only look for a particular person.

P-ict0, 2022-11-22:
Challenge **take out the garbage** has `control_to_trash_bin.py` that positions the robot to the trashbin and aligns the arm. Can't find any interesting use case but may be interesting.

shanki98, 2023-01-03:
Challenge receptionist has a smach state `PointAtReception`. 
I think this smach state can be generalized. 

Additionally, there is an importnat ToDo inside `find_empty_seat.py` to make seat designator into Robocup knowledge, such that the volume information of a seat can be changeable per seat