tue-robotics/tue_robocup #1268: Image classification broken on robot

PetervDooren, 2023-02-21:
During testing today we found that the classification does not work.

To reproduce:
- start the robot
- start free-mode
- open a python console.
```python
from robot_smach_states.world_model import SegmentObjects
from robot_smach_states.util.designators import EntityByIdDesignator                        
from robot_skills.classification_result import ClassificationResult
from robot_smach_states.util.designators import VariableDesignator  

tabledes = EntityByIdDesignator(hero, 'dinner_table')
classdesignator = VariableDesignator([], resolve_type=[ClassificationResult]) 

sm = SegmentObjects(hero, classdesignator.writeable, tabledes)
sm.execute()
```

The full error message reads: 
2023-02-21 21:16:28.564524: E tensorflow/stream_executor/cuda/cuda_dnn.cc:373] Could not create cudnn handle: CUDNN_STATUS_NOT_INITIALIZED
2023-02-21 21:16:28.564582: E tensorflow/stream_executor/cuda/cuda_dnn.cc:382] Possibly insufficient driver version: 520.61.5
2023-02-21 21:16:28.564631: W tensorflow/core/framework/op_kernel.cc:1745] OP_REQUIRES failed at conv_ops.cc:1120 : UNIMPLEMENTED: DNN library is not found.
[ERROR][/hero/object_recognition/object_recognition][1677010588.786848]: Error processing request: ('Failed to run tensorflow session: %s', UnimplementedError())
['Traceback (most recent call last):\n', '  File "/home/amigo/.local/lib/python3.8/site-packages/tensorflow/python/client/session.py", line 1377, in _do_call\n    return fn(*args)\n', '  File "/home/amigo/.local/lib/python3.8/site-packages/tensorflow/python/client/session.py", line 1360, in _run_fn\n    return self._call_tf_sessionrun(options, feed_dict, fetch_list,\n', '  File "/home/amigo/.local/lib/python3.8/site-packages/tensorflow/python/client/session.py", line 1453, in _call_tf_sessionrun\n    return tf_session.TF_SessionRun_wrapper(self._session, options, feed_dict,\n', 'tensorflow.python.framework.errors_impl.UnimplementedError: 2 root error(s) found.\n  (0) UNIMPLEMENTED: DNN library is not found.\n\t [[{{node conv/Conv2D}}]]\n\t [[final_result/_3]]\n  (1) UNIMPLEMENTED: DNN library is not found.\n\t [[{{node conv/Conv2D}}]]\n0 successful operations.\n0 derived errors ignored.\n', '\nDuring handling of the above exception, another exception occurred:\n\n', 'Traceback (most recent call last):\n', '  File "/home/amigo/ros/noetic/system/src/image_recognition_tensorflow/src/image_recognition_tensorflow/object_recognizer.py", line 70, in classify\n    predictions = sess.run(self._output_tensor,\n', '  File "/home/amigo/.local/lib/python3.8/site-packages/tensorflow/python/client/session.py", line 967, in run\n    result = self._run(None, fetches, feed_dict, options_ptr,\n', '  File "/home/amigo/.local/lib/python3.8/site-packages/tensorflow/python/client/session.py", line 1190, in _run\n    results = self._do_run(handle, final_targets, final_fetches,\n', '  File "/home/amigo/.local/lib/python3.8/site-packages/tensorflow/python/client/session.py", line 1370, in _do_run\n    return self._do_call(_run_fn, feeds, fetches, targets, options,\n', '  File "/home/amigo/.local/lib/python3.8/site-packages/tensorflow/python/client/session.py", line 1396, in _do_call\n    raise type(e)(node_def, op, message)  # pylint: disable=no-value-for-parameter\n', 'tensorflow.python.framework.errors_impl.UnimplementedError: Graph execution error:\n\nDetected at node \'conv/Conv2D\' defined at (most recent call last):\n    File "/home/amigo/ros/noetic/system/src/image_recognition_tensorflow/scripts/object_recognition_node", line 114, in <module>\n      recognizer = ObjectRecognizer(_graph_path, _labels_path, _input_tensor, _output_tensor)\n    File "/home/amigo/ros/noetic/system/src/image_recognition_tensorflow/src/image_recognition_tensorflow/object_recognizer.py", line 32, in __init__\n      _ = tf.import_graph_def(graph_def, name=\'\')\nNode: \'conv/Conv2D\'\nDetected at node \'conv/Conv2D\' defined at (most recent call last):\n    File "/home/amigo/ros/noetic/system/src/image_recognition_tensorflow/scripts/object_recognition_node", line 114, in <module>\n      recognizer = ObjectRecognizer(_graph_path, _labels_path, _input_tensor, _output_tensor)\n    File "/home/amigo/ros/noetic/system/src/image_recognition_tensorflow/src/image_recognition_tensorflow/object_recognizer.py", line 32, in __init__\n      _ = tf.import_graph_def(graph_def, name=\'\')\nNode: \'conv/Conv2D\'\n2 root error(s) found.\n  (0) UNIMPLEMENTED: DNN library is not found.\n\t [[{{node conv/Conv2D}}]]\n\t [[final_result/_3]]\n  (1) UNIMPLEMENTED: DNN library is not found.\n\t [[{{node conv/Conv2D}}]]\n0 successful operations.\n0 derived errors ignored.\n\nOriginal stack trace for \'conv/Conv2D\':\n  File "/home/amigo/ros/noetic/system/src/image_recognition_tensorflow/scripts/object_recognition_node", line 114, in <module>\n    recognizer = ObjectRecognizer(_graph_path, _labels_path, _input_tensor, _output_tensor)\n  File "/home/amigo/ros/noetic/system/src/image_recognition_tensorflow/src/image_recognition_tensorflow/object_recognizer.py", line 32, in __init__\n    _ = tf.import_graph_def(graph_def, name=\'\')\n  File "/home/amigo/.local/lib/python3.8/site-packages/tensorflow/python/util/deprecation.py", line 548, in new_func\n    return func(*args, **kwargs)\n  File "/home/amigo/.local/lib/python3.8/site-packages/tensorflow/python/framework/importer.py", line 403, in import_graph_def\n    return _import_graph_def_internal(\n  File "/home/amigo/.local/lib/python3.8/site-packages/tensorflow/python/framework/importer.py", line 516, in _import_graph_def_internal\n    _ProcessNewOps(graph)\n  File "/home/amigo/.local/lib/python3.8/site-packages/tensorflow/python/framework/importer.py", line 247, in _ProcessNewOps\n    for new_op in graph._add_new_tf_operations(compute_devices=False):  # pylint: disable=protected-access\n  File "/home/amigo/.local/lib/python3.8/site-packages/tensorflow/python/framework/ops.py", line 3926, in _add_new_tf_operations\n    new_ops = [\n  File "/home/amigo/.local/lib/python3.8/site-packages/tensorflow/python/framework/ops.py", line 3927, in <listcomp>\n    self._create_op_from_tf_operation(c_op, compute_device=compute_devices)\n  File "/home/amigo/.local/lib/python3.8/site-packages/tensorflow/python/framework/ops.py", line 3809, in _create_op_from_tf_operation\n    ret = Operation(c_op, self)\n  File "/home/amigo/.local/lib/python3.8/site-packages/tensorflow/python/framework/ops.py", line 2175, in __init__\n    self._traceback = tf_stack.extract_stack_for_node(self._c_op)\n\n', '\nDuring handling of the above exception, another exception occurred:\n\n', 'Traceback (most recent call last):\n', '  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 633, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/amigo/ros/noetic/system/src/image_recognition_tensorflow/scripts/object_recognition_node", line 67, in _recognize_srv_callback\n    predictions = self._recognizer.classify(self._bgr_image)\n', '  File "/home/amigo/ros/noetic/system/src/image_recognition_tensorflow/src/image_recognition_tensorflow/object_recognizer.py", line 74, in classify\n    raise Exception("Failed to run tensorflow session: %s", e)\n', "Exception: ('Failed to run tensorflow session: %s', UnimplementedError())\n"]
[ERROR][/hero/ed][1677010588.788590244]: Service call failed: service [/hero/object_recognition/recognize] responded with an error: error processing request: ('Failed to run tensorflow session: %s', UnimplementedError())
[ERROR][/hero/ed][1677010588.788653084]: [ED Perception] Classification service call failed

PetervDooren, 2023-02-21:
@shanki98 was here as well

MatthijsBurgh, 2023-02-23:
I think you can trigger this by using the rat test plugin and selecting the correct image topic and classification service. Easier to test IMO.

MatthijsBurgh, 2023-03-03:
I am pretty sure, somebody broke the installation. As in CI it still works correctly and tensorflow hasn't been updated since november.

MatthijsBurgh, 2023-03-04:
Installation was indeed broken. Fixed it.

closed 2023-03-04 by MatthijsBurgh

PetervDooren, 2023-03-04:
@MatthijsBurgh for future reference. The installation of what exactly was broken and what should someone look at to fix it?

MatthijsBurgh, 2023-03-04:
I think it was related nvidia/cuda stuff. I deleted a bunch of stuff and reinstalled stuff.